ros-perception/pointcloud_to_laserscan #5: Support for ROS Kinetic

mervynwee, 2016-08-16:
Hi, is there a release version for jade and kinetic? Thanks!

owner01, 2016-10-26:
You can easily compile it just put it in the src directory of a catkin_ws and use catkin_make. Then just source devel/setup.bash and ou will be able to run it. Unfortunately i currently only get 1 frame every 15 seconds but don't know if this is due to me or somethign wrong in the code I will keep you updated

tfoote, 2016-10-26:
Note that if you're using a depth camera it's usually more efficient to use [depthimage_to_laserscan](http://wiki.ros.org/depthimage_to_laserscan) and has replaced the usage of this in most demos.

paulbovbel, 2016-10-26:
https://github.com/ros/rosdistro/pull/13069#event-837777788 released, thanks for holding

closed 2016-10-26 by paulbovbel