ros-perception/pointcloud_to_laserscan #9: high cpu usage

zhengzh, 2017-05-03:
tf2_->transform(*cloud_msg, *cloud, target_frame_, ros::Duration(tolerance_));
high cpu usage -_-

jonbinney, 2017-05-03:
How many points are in the point cloud you are transforming? How often are you transforming it? What CPU are you on? What ROS distro are you on?

zhengzh, 2017-05-03:
@jonbinney 640*480 points, ubuntu 16.04, kinect v1 (30fps), thinkpad t440p (i7 cpu)
the fps drop down to about 0.6 fps When executing this line of code
30(fps) when i comment it out.
I don't know why tf2_->transform is so slow

paulbovbel, 2017-05-03:
Tf2 will transform every single point in the cloud. I've found it helpful to run the point cloud through a PCL voxel filter first.

zhengzh, 2017-05-03:
@paulbovbel thanks! I need to transform the point cloud from kinect_link to base_link,  i will try pcl library,

closed 2017-05-03 by paulbovbel