ros-perception/pointcloud_to_laserscan #13: Error pluginlib::InvalidXMLException

soutieslak, 2018-01-12:
When running a the following command: roslaunch pointcloud_to_laserscan sample_node.launch

I get this error:

process[pointcloud_to_laserscan-16]: started with pid [28156]
terminate called after throwing an instance of 'pluginlib::InvalidXMLException'
  what():  XML Document has no Root Element.  This likely means the XML is malformed or missing.

I am running ROS Lunar. Could anybody assist me?

paulbovbel, 2018-01-12:
That's bizarre - I'm not having the same error when running from binaries. Are you building from source?

soutieslak, 2018-01-12:
Hi Paul! Yes I am. This is what I tried, maybe you can point out a step I am missing, or where I went wrong. 

1. git clone in my catkin_ws/src
2. catkin_make 
3. source devel/setup.sh
4. roslaunch pointcloud_to_laserscan sample_node.launch

And then I get that issue when starting the node... 

My LiDAR node runs fine, noticed this error happens with it running/or not running.

Am I missing a setting that I should set?

I first came about this error when I tried using the scanse/sweep-ros launch file called "sweep2scan.launch which calls the pointcloud_to_laserscan package. This is when I first got the error, and then tested it out seperatley and got the same issue. (here is the scanse git repository:  https://github.com/scanse/sweepers )

Any help would be great man!

tfoote, 2018-01-12:
@soutieslak What platform are you running on? What version of pointcloud_to_laserscan and pluginlib are you running? Can you test from binary packages only w/o your workspace?

soutieslak, 2018-01-15:
@tfoote thank your for the reply. I am using ROS Lunar, on Ubuntu 17.04

I am not sure which version of pointcloud_to_laserscan I am using, but when I initially got the error, I removed it (sudo apt-get remove ros-lunar-pointcloud-to-laserscan) and then I added it again (sudo apt-get install ros-lunar-pointcloud-to-laserscan) and when that still gave the same error, I then performed a "git clone" of this version here. The package.xml file reads V1.4.0

The pluginlib package.xml file reads V1.11.0

When I call the package without being in my workspace I get the same error.

Thanks again for your replies and assistance, I appreciate your help

soutieslak, 2018-01-16:
When bypassing the launch file and just trying to run the pointcloud_to_laserscan on the pointcloud_to_laserscan_node I get the same error:

[image]

Is it possible that my pluginlib is damaged/corrupt?

soutieslak, 2018-01-16:
I just setup another PC with ROS, same config as mine and it works there... So that means there is no issue with the package... must be something weird with my PC's setup... Nooooooooo!!!! 

Any suggestions as to what it could possibly be? Would like to try avoid formatting... Is there a know "fix ros install" that you guys maybe know about?

mikaelarguedas, 2018-01-17:
> same config as mine and it works there...

Can you confirm what version of pluginlib you're running on both machines?
`apt-cache show ros-lunar-pluginlib | grep Version`

The last released version 1.11.2 includes a fix to workaround the issue of unreleated malformed xml files throwing errors.

Another issue we faced (not sure if it's related) is that tinyxml2 on Ubuntu Zesty is broken so we had to release a patched version in the ROS repositories. Can you please make sure that you have the version of tinyxml2 distributed by ROS (4.0.1-2zesty and **not** 4.0.1-1):
```
$ apt-cache show libtinyxml2-dev | grep Version
Version: 4.0.1-2zesty # this is the ROS version
```

soutieslak, 2018-01-17:
I will have a look on the other machine now now, but for now this is what it looks like on the machine giving the error:

[image]

Is it normal that it lists 2 versions of both?

tfoote, 2018-01-17:
@soutieslak please copy and paste console output contents instead of screenshotting. That way text is copyable, searchable, and not potentially blurry.

soutieslak, 2018-01-17:
Thanks @tfoote, I will do so in the future

mikaelarguedas, 2018-01-20:
@soutieslak It looks like your issue is due to the fact that gmapping 3.9 is installed on one machine and not the other (see https://github.com/ros-perception/slam_gmapping/pull/56 for more details).

Can you confirm that removing gmapping package fixes your issue?
```
sudo apt remove ros-lunar-gmapping
```  

Though pluginlib 1.11.2 should fix the issue...
The error message should still be displayed but `pointcloud_to_laserscan` should not crash. Can you please confirm that you still have `Aborted (core dumped)` on the console of the machine that's failing?

On my side I have something like this:
```
root@830f21cbbf76:/# rosrun pointcloud_to_laserscan pointcloud_to_laserscan_node 
[ERROR] [1516468010.020242179]: Skipped loading plugin with error: XML Document '/opt/ros/lunar/share/gmapping/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1516468010.021229441]: Initializing nodelet with 8 worker threads.
```
Note that `pointcould_to_laserscan` is running correctly after printing the error message

mikaelarguedas, 2018-01-24:
I believe this can be closed now that https://github.com/ros-perception/slam_gmapping/pull/56 has been merged ans released

closed 2018-01-24 by paulbovbel