ros-perception/pointcloud_to_laserscan #16: Changing the default camera name result in no laserscan published

TheRobotStudio, 2018-04-26:
Hi, when I launch the sample_node.launch with camera arg as "xtion", it all populates at the right places but I have no laserscan published, am I missing something ?
Otherwise with the default "camera" name everything works fine, I can see my laserscan in Rviz.
Thanks!

wangguojun2018, 2018-06-21:
Do you solve the problem

TheRobotStudio, 2018-06-21:
@wangguojun2018 No, I haven't look at it since I posted that issue. If I find a solution I'll publish it here.

paulbovbel, 2018-06-25:
Changing the camera name means making sure your new camera publishes these same topics:

```
        <remap from="cloud_in" to="$(arg camera)/depth_registered/points_processed"/>
        <remap from="scan" to="$(arg camera)/scan"/>
```

It's possible that the openni2 stack has changed since I initially  created those launch files, but I no longer have access to a sensor to test. Feel free to verify, adapt, or PR a fix for the sample launches.

closed 2018-06-25 by paulbovbel