ros-perception/pointcloud_to_laserscan #27: laserscan_to_pointcloud package

reinzor, 2019-04-29:
Not sure whether I am at the right address here but wouldn't it be nice to also add a `laserscan_to_pointcloud` package? I could do the initial package here but I do not have any rights in the perception group.

paulbovbel, 2019-04-29:
Hi @reinzor, this kind of thing can be accomplished via http://wiki.ros.org/laser_filters#scan_to_cloud_filter_chain. The main downside is that laser_filters doesn't offer a nodelet implementation, meaning that it can't be used as part of a nodelet zero-copy pipeline for scans/pointcloud.

Generally a PR implementing a laserscan_to_pointcloud nodelet in this package would be welcome.

tfoote, 2019-04-30:
There's a library for this in `laser_geometry` http://wiki.ros.org/laser_geometry

reinzor, 2019-04-30:
Thanks, I am aware of the `laser_geometry` package. I am working on a nodelet implementation that can be added in this package.

reinzor, 2019-05-01:
Closed by https://github.com/ros-perception/pointcloud_to_laserscan/issues/28

closed 2019-05-01 by reinzor