ros-perception/pointcloud_to_laserscan #32: scan topic is empty

walid19, 2019-07-23:
I am trying to convert Pointcloud2 msg  to laser scan, I mapped the input topic in the launch file . the out laser scan topic is empty ?.

kungfrank, 2019-11-11:
Hello, I have the same problem. Did you solve it?

walid19, 2019-11-11:
Hi not yet I am still struggling, plz if you fix it let me know

hynkis, 2019-11-20:
1. Try to run this command in terminal at first.
rosrun pointcloud_to_laserscan pointcloud_to_laserscan_node cloud_in:=<your pointcloud topic>
and subscribe topic "/scan" (/scan is the default publish topic).

2. If it works, then it means pointcloud_to_laserscan package works and you need to change the parameters in the example launch file (simple_node.launch). In my case, "scan_time" parameter was too big(default 0.33, mine was 0.033). You can check the right parameters by 'rostopic echo' the message "/scan" when you run the 'rosrun' command above.

walid19, 2019-11-20:
Thank you so much it works for me now :)

closed 2019-11-20 by SteveMacenski