ros-perception/pointcloud_to_laserscan #34: 将点云转换为/camera/scan后和激光雷达/scan进行融合，用于gmapping出现问题

anchuanxu, 2020-03-20:
### 描述

您好，我利用这个软件包将相机的点云数据成功转换成了/camera/scan，并且可以利用该话题进行gmapping建图，但是当我将/camera/scan和激光雷达原始的/scan两个数据进行融合，企图将悬空物体建出地图却失败了，请问你有可能知道是什么原因吗？

#### 想要建出的图

[image]

#### 实际融合之后建出的图

[image]

#### 点云情况

[image]

可以看到，悬空物体的后面激光雷达数据已经被切割，但是为何还是建不出想要的边，地图上显示的还是没有。谢谢

Dennis174698, 2020-09-02:
你好，qing

> ### 描述
> 您好，我利用这个软件包将相机的点云数据成功转换成了/camera/scan，并且可以利用该话题进行gmapping建图，但是当我将/camera/scan和激光雷达原始的/scan两个数据进行融合，企图将悬空物体建出地图却失败了，请问你有可能知道是什么原因吗？
> 
> #### 想要建出的图
> [image]
> 
> #### 实际融合之后建出的图
> [image]
> 
> #### 点云情况
> [image]
> 
> 可以看到，悬空物体的后面激光雷达数据已经被切割，但是为何还是建不出想要的边，地图上显示的还是没有。谢谢

你好，请问你实现的步骤是什么样的呢？

SteveMacenski, 2020-09-02:
You may find maintainers are more responsive if you converse in English (as the majority of the maintainers are American). While I encourage all to participate in whatever way they’re able to, technical details can easily be lost or misunderstood through automated translators if you happen to know some english.

Dennis174698, 2020-09-03:
你好，请问您是设置的步骤是怎么样的呢？

Kevinnan-teen, 2020-11-09:
gmapping只能建立2D地图

closed 2024-09-11 by paulbovbel