ros-perception/pointcloud_to_laserscan #35: Scan output still contains inf even when use_inf is false

hughcorley, 2020-04-05:
Data on the `scan` topic still contains inf data points even when `use_inf` is false. To remedy this, `range_max` must be set to something, and `use_inf` must be false. In other words, the default value of `range_max` does not seem to affect the infinite range (defined as `range_max` + 1 if `use_inf` is false). Instead, `range_max` must explicitly be given a value in order for the infinite range to be `range_max` + 1.

paulbovbel, 2020-04-05:
`range_max` is double max by default (https://github.com/ros-perception/pointcloud_to_laserscan/blob/lunar-devel/src/pointcloud_to_laserscan_nodelet.cpp#L70). Where do you see it translated into `inf`?

hughcorley, 2020-04-06:
Sorry maybe I wasn't clear. I don't mean to say `range_max` is ever `inf`. What I mean to say is that setting `use_inf` to false does not work unless you also explicitly set `range_max` to something. Does that make sense?

paulbovbel, 2020-04-06:
I wasn't clear either :)

`range_max` drives a few behaviours:
- it's the default value of any ray when `!use_inf`
- any trace larger than `range_max` is ignored

So if `!use_inf`, and you're seeing `inf` reported in the laserscan, then it must be driven by range_max. Can you provide your launch file?

hughcorley, 2020-04-07:
Not to worry :)

Right, so I set `!use_inf` and if I do not explicitly set `range_max` (i.e. I let it use its default value) then I see `inf` in the data. In other words, the following launch code results in data which is upper bounded by some finite value
```
    <node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">
      <remap from="cloud_in" to="velodyne_points"/>
      <param name="use_inf" value="false"/>
      <param name="range_max" value="200"/>
    </node>
```
while this does not
```
    <node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">
      <remap from="cloud_in" to="velodyne_points"/>
      <param name="use_inf" value="false"/>
    </node>
```

That make sense? 

Thank you for the prompt replies! Hope you stay well during this time.

closed 2026-11-02 by Rayman