ros-perception/pointcloud_to_laserscan #39: Getting a curved point cloud and no laserscan for flat wall

KaranManghi, 2020-05-27:
Hi,
   I am getting a curved point cloud for some reason for some scenes even though there are no curved objects. I used the exact same parameters as shown in the pointcloud_to_laserscan_node.cpp. I am using a RealSense camera. Also, when pointed to a wall, it does not show any laserscan. The wall might be at a distance of a meter. 
Kindly help me...

Thank you..

Regards,
Karan Manghi

paulbovbel, 2020-06-05:
Hi @KaranManghi, this kind of thing might be better suited for answers.ros.org, since it's not a bug. You should post your configuration there. Feel free to link to the question here, but I'll close the issue in the meantime.

closed 2020-06-05 by paulbovbel

KaranManghi, 2020-06-24:
Got it.... Thanks @paulbovbel