ros-perception/pointcloud_to_laserscan #44: Document node subscription behaviour

mortenfyhn, 2020-09-11:
Hi all!

I tried running the pointcloud to laserscan node, by launching the node and remapping `cloud_in` to my pointcloud topic. I then checked the rqt node graph to see if everything looks nice and connected. But it doesn't! I did some digging and realized that the you must subscribe to the laserscan output before the node will subscribe to the pointcloud input. This was confusing for me. Can we document this behaviour somewhere? I suppose the [wiki page](https://wiki.ros.org/pointcloud_to_laserscan) is a good location. I tried editing it myself, but I'm having issues with my ros.org login. Another, maybe better way is to have the node print something like "waiting for someone to subscribe to <topic>" when starting if nobody subscribes right away, then it will be very clear to the user what he needs to do.

mortenfyhn, 2020-09-11:
I noticed by chance now that this is indeed documented in a repository readme added for the v2 branch. Maybe we could add a similar readme to the current default branch.

paulbovbel, 2020-09-11:
I've updated the default branch to point at ROS2 (onwards!), but PRs are welcome to update the ROS1 readmes

mortenfyhn, 2020-09-12:
Cool! I think that in itself pretty much solves this, because now ROS1 users will likely notice the readme.

hidmic, 2020-12-09:
@mortenfyhn I'll assume you won't be submitting PRs to update ROS1 READMEs. I'll go ahead and close this issue then. Feel free to re-open if need be.

closed 2020-12-09 by hidmic