ros-perception/pointcloud_to_laserscan #50: Misleading documentation on the "scan_time" parameter

fnilius, 2021-01-29:
Hi,

First of all, thanks a lot for this great package!

There is one point I find quite misleading in the documentation : the explanation of the "scan_time" parameter.

On the wiki ([http://wiki.ros.org/pointcloud_to_laserscan#Parameters](http://wiki.ros.org/pointcloud_to_laserscan#Parameters) : 

> ~scan_time (double, default: 1.0/30.0)
> 
>     The scan rate in seconds. 

And on the readme on the git : 

> scan_time (double, default: 1.0/30.0) - The scan rate in seconds. Used to populate the output laser scan.

The first definitions leads to think that this parameter enables to down-sample the frequency of the output messages.

The second definition is quite unclear, weather it is the down-sampling the output messages or just for filling the messages.

After asking on [Ros Answers](https://answers.ros.org/question/370589/pointcloud_to_laserscan-publish-rate-higher-than-scan_time/) it seems that the scan_time is only put into the corresponding field of the output messages, and not used for any computations.

If the conclusions are correct, I would suggest changing the message to a clearer one, for example : 

> scan_time (double, default: 1.0/30.0) - The scan rate in seconds. Only used to populate the scan_time field of the output laser scan message.

Thanks a lot in advance

Best regards

Felix

hidmic, 2021-02-08:
> After asking on Ros Answers it seems that the scan_time is only put into the corresponding field of the output messages, and not used for any computations.

Indeed. Would you be willing to contribute such change @fnilius ?

paulbovbel, 2021-02-08:
Let's not drag this out, changed README - thanks for the suggestion @fnilius, feel free to PR directly next time!

closed 2021-02-08 by paulbovbel