ros-perception/pointcloud_to_laserscan #53: Where can I find this for ROS1?

konnoeric, 2021-04-02:
I came here from ROS tutorial. Where can I find pointcloud_to_laserscan for ROS1?

paulbovbel, 2021-04-02:
Which tutorial?

This is the latest ROS1 branch https://github.com/ros-perception/pointcloud_to_laserscan/tree/lunar-devel

closed 2021-04-02 by paulbovbel

konnoeric, 2021-04-06:
I came from this website. [http://wiki.ros.org/pointcloud_to_laserscan](url)

I git cloned this latest ROS1 branch [https://github.com/ros-perception/pointcloud_to_laserscan/tree/lunar-devel](url) and did catkin_make but this error below showed. Where can I find a package that I can use it on ROS1?

  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/project/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/project/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

hidmic, 2021-04-06:
@konnoeric your issues are unrelated to this package. I'd encourage you to revisit ROS documentation, particularly documentation on [building packages](http://wiki.ros.org/ROS/Tutorials/BuildingPackages).