ros-perception/pointcloud_to_laserscan #60: Output laserscan cannot be subscribed to properly

CFZeon, 2021-09-22:
Hello! 

Thank you for making this package. However, I am having issues with subscribing to the output laser scans. 
When I launch sample_pointcloud_to_laserscan_launch.py and subscribe to it in RViz, it shows Showing [0] points from [0] messages and the node indicates that it is not processing any input due to a lack of subscribers. 
However, ros2 topic echo counts as a valid subscriber and when I wrote a sample node to subscribe to it, it does not work as well.

Do you know why this happens?

Regards,
Zeon

hidmic, 2021-10-19:
@CFZeon IIUC what you describe has less to do with `pointcould_to_laserscan` than with proper pub-sub, which makes this a non-issue. I don't if you're familiar with http://answers.ros.org/, but we typically encourage folks to ask questions there instead. That increases the likelihood of you getting a prompt response, and the broader community can benefit from that exchange. Thus, I'll proceed to close this issue.

> Do you know why this happens?

I'd verify you have matching [QoS settings](https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html), so start with.

closed 2021-10-19 by hidmic

qzsrh, 2023-11-15:
I also had the same problem, please how did you solve this problem in the end

DarK404, 2024-04-24:
It should be fixed by setting up the same the **reliability** in my case 
use `ros2 topic info /scan --verbose
`to check the [QoS settings](https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html), already used.

you can edit the code like this! 
```
pub_ = this->create_publisher<sensor_msgs::msg::LaserScan>("scan", rclcpp::SensorDataQoS());
```
```
pub_ = this->create_publisher<sensor_msgs::msg::LaserScan>("scan", rclcpp::SensorDataQoS().reliability(RMW_QOS_POLICY_RELIABILITY_RELIABLE));
```