ros-perception/pointcloud_to_laserscan #61: Package failed with error "C2065: 'M_PI': undeclared identifier"

XavierTrudeau, 2021-09-28:
I was trying to build this package and was getting "Package failed" with no error message.
I'm using ROS2 Foxy on Windows 10 with Visual Studio 2019.

By using this command while building:
colcon build --packages-select pointcloud_to_laserscan --event-handlers console_cohesion+

I was able to find this error : 
"error C2065: 'M_PI': undeclared identifier [C:\nav2_ws\build\pointcloud_to_laserscan\pointcloud_to_laserscan.vcxproj]"

I was able to build it by adding this line to "pointcloud_to_laserscan/pointcloud_to_laserscan_node.cpp" :
#define M_PI 3.14159265358979323846

I hope this helps. I'm not sure if that was the right approach as I'm quite new to ROS!

Regards,
Xavier Trudeau

kenji-miyake, 2021-09-29:
Probably it's because the behavior of `_USE_MATH_DEFINES` is different between gcc and msvc? :thinking: 
https://docs.microsoft.com/en-us/cpp/c-runtime-library/math-constants?view=msvc-160

XavierTrudeau, 2021-09-29:
That's right. Adding these lines to the header file also fixed it.
#define _USE_MATH_DEFINES // for C++
#include <cmath>

I'm now getting a crash when trying to run the sample launch file. Both pointcloud_to_laserscan_node and laserscan_to_pointcloud_node crash upon launch : 

[ERROR] [pointcloud_to_laserscan_node.EXE-3]: process has died [pid 7072, exit code 3221226505, cmd 'c:\nav2_ws\install\lib\pointcloud_to_laserscan\pointcloud_to_laserscan_node.EXE --ros-args -r __node:=pointcloud_to_laserscan -r __node:=pointcloud_to_laserscan --params-file C:\Users\xaviert\AppData\Local\Temp\launch_params_45izntrr -r cloud_in:=scanner/cloud -r scan:=scanner/scan'].

Any idea what could cause this?

hidmic, 2021-10-13:
@XavierTrudeau a backtrace would surely help.

closed 2024-09-11 by paulbovbel