ros-perception/pointcloud_to_laserscan #81: CMake Error at CMakeLists.txt  when colcon build with Galactic

patrickpoirier51, 2023-08-11:
Hello
I get this error when building package

colcon build --packages-select pointcloud_to_laserscan

Starting >>> pointcloud_to_laserscan
--- stderr: pointcloud_to_laserscan                         
CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:442: cmake_check_build_system] Error 1
---
Failed   <<< pointcloud_to_laserscan [0.96s, exited with code 2]

I am running with:
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.5 LTS"

cmake version 3.16.3
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

vico1999-ros, 2023-11-16:
I find the the same error tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found. Perhaps a
find_package() call is missing for an IMPORTED target, or an ALIAS target
is missing?            but the another package didnot induce libary error

jackkim777, 2023-11-26:
me either. i got error like this (and i use foxy)
Starting >>> pointcloud_to_laserscan
--- stderr: pointcloud_to_laserscan                         
CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:442: cmake_check_build_system] Error 1
---
Failed   <<< pointcloud_to_laserscan [2.22s, exited with code 2]

Summary: 0 packages finished [5.19s]
  1 package failed: pointcloud_to_laserscan
  1 package had stderr output: pointcloud_to_laserscan

AnnaPlt, 2024-02-12:
@patrickpoirier51 Hello! I am encountering the same errors on the same packages. Did you find a solution?

patrickpoirier51, 2024-02-12:
did you tried the package  ros-galactic-depthimage-to-laserscan

NemB0t, 2024-02-23:
I am also facing the same issue any ideas on how to fix it or has anyone found any work arounds?

Lem-bow, 2024-03-22:
@NemB0t Did you checkout the correct branch after cloning? That did the job for me.

Merlinecodes, 2024-04-18:
Please try this:
git clone [url] --branch=[branch name]

I stopped getting the CMake error once I cloned the correct branch (foxy in my case)

NemB0t, 2024-08-05:
It worked after @Merlinecodes suggestions. Thanks!

closed 2024-09-11 by paulbovbel