ros-perception/pointcloud_to_laserscan #86: colcon build error

vico1999-ros, 2023-11-16:
colcon build --packages-select pointcloud_to_laserscan
Starting >>> pointcloud_to_laserscan
--- stderr: pointcloud_to_laserscan                         
CMake Error at CMakeLists.txt:23 (target_link_libraries):
  Target "laserscan_to_pointcloud" links to:

    tf2_sensor_msgs::tf2_sensor_msgs

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Error at CMakeLists.txt:44 (target_link_libraries):
  Target "pointcloud_to_laserscan" links to:

    tf2_sensor_msgs::tf2_sensor_msgs

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
   I checked my UBUNUTU 22.04 ROS2 have installed the package .
(base) vico@vico-Robo:~$ ros2 pkg prefix tf2_sensor_msgs
/opt/ros/humble
(base) vico@vico-Robo:~$ echo $PATH
/usr/local/rs_driver/include/rs_driver/:/usr/lib/pycharm-community-2023.1.3/bin:/usr/local/cuda/bin:/home/vico/miniconda3/bin:/home/vico/miniconda3/condabin:/opt/ros/humble/bin:/home/vico/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
(base) vico@vico-Robo:~$ export PATH=/opt/ros/humble:$PATH

vico1999-ros, 2023-11-16:
i find the packages cmakelist is not suitable for the code ,  need to update with the ROS2 version。 i replace it and find it can work now

closed 2023-11-16 by vico1999-ros

luj97, 2023-12-14:
@vico1999-ros What did you change in the cmakelist?