ros-perception/pointcloud_to_laserscan #87: New pointcloud_to_lasersacan release on Rolling

ahcorde, 2024-02-08:
@wep21 Do you mind to release on Rolling ?
CI is failing [herehttps://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__pointcloud_to_laserscan__ubuntu_jammy_amd64__binary/114/console(https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__pointcloud_to_laserscan__ubuntu_jammy_amd64__binary/114/console) because I removed some deprecated headers https://github.com/ros2/geometry2/pull/645 in geometry2

wep21, 2024-02-08:
@ahcorde I'm sorry, but I neither have write access to this repo, nor belong to [release team](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/f1a07316cd96f8c194959dfbdcc5e21c63926934/perception.tf#L3-L7).
If someone gives me the right, I can do that.

paulbovbel, 2024-09-11:
Looks like this was done, thanks!

https://github.com/ros-perception/pointcloud_to_laserscan/commit/d71ed2ea7405249bc5f0b2e15b89434e3291b1ce

closed 2024-09-11 by paulbovbel