ros-perception/pointcloud_to_laserscan #88: Issue with colcon build in ros2 foxy

fatemehprhm, 2024-02-14:
oprating system: ubuntu 20.04
ros version: foxy
when i try to build the library by ``` colcon build --packages-select pointcloud_to_laserscan``` i encounter this error:
```
Starting >>> pointcloud_to_laserscan
--- stderr: pointcloud_to_laserscan                         
CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "rclcpp_components::component" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:266: cmake_check_build_system] Error 1
---
Failed   <<< pointcloud_to_laserscan [1.81s, exited with code 2]

Summary: 0 packages finished [7.24s]
  1 package failed: pointcloud_to_laserscan
  1 package had stderr output: pointcloud_to_laserscan```
  Does anybody know how can i solve it?

TZECHIN6, 2024-06-27:
Remember to switch correct branch before building project

closed 2024-09-11 by paulbovbel

Adwaithottonomy, 2025-06-05:
clone with this command -> "git clone -b foxy https://github.com/username/repository_name.git"