ros-perception/pointcloud_to_laserscan #89: issue with colcon build in ros2 humble

uo278193, 2024-02-19:
ros version:humble
Starting >>> pointcloud_to_laserscan
--- stderr: pointcloud_to_laserscan                           
CMake Error at CMakeLists.txt:15 (add_library):
  Target "laserscan_to_pointcloud" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:36 (add_library):
  Target "pointcloud_to_laserscan" links to target
  "tf2_sensor_msgs::tf2_sensor_msgs" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< pointcloud_to_laserscan [12.5s, exited with code 1]
                                 
Summary: 0 packages finished [16.9s]
  1 package failed: pointcloud_to_laserscan
  1 package had stderr output: pointcloud_to_laserscan

alvikag, 2024-02-22:
I am facing the same error on galactic. any luck?

dldnguyen-fr, 2024-04-02:
Did you guys make sure to pull the correct branch for your ROS2 version?

Merlinecodes, 2024-04-18:
Please try this: 
git clone [url] --branch=[branch name]

I stopped getting the CMake error once I cloned the correct branch and compiled it (foxy in my case)

p-hoanganh, 2024-05-23:
Hi,
You can try:
git clone -b humble [url]
It works for me.

TZECHIN6, 2024-06-27:
Remember to switch to correct branch for your ros distro.

```
# see which branch is available for your ros distro version
git branch -a
```
```
# switch to correct branch before build
git switch <branch_name>
```

closed 2024-09-11 by paulbovbel