ros-perception/pointcloud_to_laserscan #94: Adaptive mid360？

makanov001, 2024-10-31:
May I ask whether this algorithm is applicable to the data of livox-mid360

Un1Lee, 2024-11-12:
I have the same question. What's more, the output laserscan in rviz shows that status ok but 0 point.

Un1Lee, 2024-11-12:
> May I ask whether this algorithm is applicable to the data of livox-mid360

it can not for livox form, you can use livox code for rviz, it's pointcloud2 form. Or you change this code for livox form. My question about mid360 is its param for laserscan.

makanov001, 2024-11-12:
> > May I ask whether this algorithm is applicable to the data of livox-mid360
> 
> it can not for livox form, you can use livox code for rviz, it's pointcloud2 form. Or you change this code for livox form. My question about mid360 is its param for laserscan.

I tried to use livox custom pointcloud format (livox_ros_driver2/CustomMsg) and pointcloude2 format (sensor_msgs/PointCloud2) without success, my attempt seems to be similar to your description, There is no point cloud output in rviz

Un1Lee, 2024-11-12:
> I tried to use livox custom pointcloud format (livox_ros_driver2/CustomMsg) and pointcloude2 format (sensor_msgs/PointCloud2) without success, my attempt seems to be similar to your description, [...]

have you check your rqt_graph that it really convert the cloud_in to scan? I used to make mistake in topic name. Now I have correct rqt_graph and right frame id. I echo scan topic, I can get range which seems OK. But rviz show nothing.
[image]

makanov001, 2024-11-12:
> echo scan topic

It seems that you are closer to success, I may also use the wrong topic name, when I echo scan topic, there are a lot of invalid point data (inf), and each scan data is the same, in addition, I also have a problem is that when I change the configuration parameter in launch.py, For example, after range_min or range_max, echo scan topic still displays the original configuration parameters. The following is a piece of data displayed by my echo scan topic. If possible, please provide your configuration file. I want to see if I used the wrong topic name.
---------------
header:
stamp:
sec: 1730296024
nanosec: 516303117
frame_id: cloud
angle_min: -1.5707999467849731
angle_max: 1.5707999467849731
angle_increment: 0.008700000122189522
time_increment: 0.0
scan_time: 0.33329999446868896
range_min: 0.44999998807907104
range_max: 4.0
ranges:
- .inf
- .inf
- .inf
- .inf
- .inf
- 0.9698204398155212
- .inf
- .inf
- .inf
- .inf
- .inf
- 0.7160065770149231
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- 0.9560691714286804

Un1Lee, 2024-11-12:
range_min: 0.44999998807907104
range_max: 4.0. 
Change this paramter in launch file, mid360 not only 4.0. I change it as 20. the > max_range will show as inf.
I will check the param for more exactly param tomorrow. but I think my ranges has no error now, why rviz show 0 point. I use fastlio2, which sub livox and pub pointcloud2. My project need fastlio2. So I use msg_MID. But I think rviz_MID360 is totally OK, and I think my fastlio2 output is similar with rviz_MID360.

Un1Lee, 2024-11-12:
oh!! I solve it. it's communication problems in rviz. https://github.com/ros2/rviz/issues/578
78 line, create_publish, rclcpp::SensorDataQoS()->10. build then run, rviz would show 330 points.

makanov001, 2024-11-12:
Congratulations on solving this problem, can you help me with something? I read the link you shared, but I don't quite understand which code in which file needs to be changed. In addition, could you please provide your complete launch file? I think my topic name relationship may be incorrectly configured. Thank you.

Un1Lee, 2024-11-13:
> Congratulations on solving this problem, can you help me with something? I read the link you shared, but I don't quite understand which code in which file needs to be changed. In addition, could you [...]

All you need is /src/pointcloud_to_leaserscan_node.cpp(named node.cpp) and /launch/sample_pointcloud_to_laserscan_launch.py(named launch.py). Delet all remapping in launch.py, so that your topic name only depends on node.cpp. In node.cpp,  78 line is publich, and 123 line is subscribe, change the publich name as your pointcloud2 topic, and change the subscribe name as you want. Then you can check this in rqt_graph.

rviz status show ok but no point is beause communication problems. To address it, you only need to change code in 78 line, the public in node.cpp.  Change the pub frequen: rclcpp::SensorDataQoS()->10. Then the line 78 line should be like: pub_ = this->create_publisher<sensor_msg::msg::LaserScan>('/scanner/scan/', 10)

makanov001, 2024-11-13:
According to the solution you gave, I successfully solved this problem, thank you for sharing!
[image]

maxxbolotov, 2024-12-01:
@makanov001 or may be you can provide the step by step tutorial please?

makanov001, 2024-12-02:
Hello, @maxxbolotov ,you need to change the relevant parameters of two files, the two files are: /src/pointcloud_to_leaserscan_node.cpp(named node.cpp) and /launch/sample_pointcloud_to_laserscan_launch.py(named  launch.py).

1-In launch.py, Delet all remapping in launch.py, so that your topic name only depends on node.cpp.
The modified file is as follows:
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

def generate_launch_description():
return LaunchDescription([
DeclareLaunchArgument(
name='scanner', default_value='scanner',
description='Namespace for sample topics'
),
Node(
package='tf2_ros',
executable='static_transform_publisher',
name='static_transform_publisher',
arguments=[
'--x', '0', '--y', '0', '--z', '0',
'--qx', '0', '--qy', '0', '--qz', '0', '--qw', '1',
'--frame-id', 'map', '--child-frame-id', 'cloud'
]
),
Node(
package='pointcloud_to_laserscan', executable='pointcloud_to_laserscan_node',
parameters=[{
'target_frame': 'cloud',
'transform_tolerance': 0.01,
'min_height': 0.0,
'max_height': 1.0,
'angle_min': -3.14, # -M_PI/ 2-1.5708
'angle_max': 3.14, # M_PI/2
'angle_increment': 0.0087, # M_PI/360.0
'scan_time': 0.3333,
'range_min': 0.45,
'range_max': 40.0,
'use_inf': True,
inf_epsilon: 1.0
}].
name='pointcloud_to_laserscan'
)
])

2-In node.cpp, 78 line is publich, and 123 line is subscribe, change the publich name as your pointcloud2 topic,  and change the subscribe name as you want. Also, you need to change the frequency of publich on line 78 to fix the display of rviz.
The modified code is as follows:
78 line: pub_ = this->create_publisher<sensor_msgs::msg::LaserScan>("/scanner/scan",10);
123 line: sub_.subscribe(this, "/livox/lidar", qos.get_rmw_qos_profile());

3- Data issues
If you are using livox_ros_driver2 to collect data directly for testing, make sure you are using the rviz_MID360_launch.py format, not msg_MID360_launch.py

maxxbolotov, 2024-12-02:
> Hello, @maxxbolotov ,you need to change the relevant parameters of two files, the two files are: /src/pointcloud_to_leaserscan_node.cpp(named node.cpp) and [...]

I did everything as you said. Also launch rviz_MID360_launch.py but still got 0 points from 0 messages beside laserscan topic is ok. 
Also my pointcloud_to_laserscan log is: 
Message filter dropping message: frame ‘livox_frame’ at time 173313577.174 for reason ‘discarding message because the queue is full’

maxxbolotov, 2024-12-02:
Solved after changing launch.py using lidar configurations.

rtarun1, 2024-12-04:
I am facing the same issue @maxxbolotov , can you please elaborate on how to fix it? Thank you

maxxbolotov, 2024-12-04:
> I am facing the same issue @maxxbolotov , can you please elaborate on how to fix it? Thank you

Sure, here it is:
1) I’m using Fast_Lio so my topic that is publishing pointcloud is /cloud_registered_body (or something like that. You can check it in RVIZ after starting fast_lio). So also to publish pointclouds I’m using msg launch file, not rviz. 
2) As topic name is cloudr_registered_body you have to replace topic name in line 123 in cpp file of pointcloud_to_laserscan. 
3) The last thing to do is to change your launch file pointcloud_to_laserscan as makanov001 provide in the previous answers.

rtarun1, 2024-12-07:
Thank you @maxxbolotov it works now!

hermanumrao, 2025-03-14:
> > I am facing the same issue [@maxxbolotov](https://github.com/maxxbolotov) , can you please elaborate on how to fix it? Thank you
> 
> Sure, here it is:
> 
>     1. I’m using Fast_Lio so my topic that is publishing pointcloud is /cloud_registered_body (or something like that. You can check it in RVIZ after starting fast_lio). So also to publish pointclouds I’m using msg launch file, not rviz.
> 
>     2. As topic name is cloudr_registered_body you have to replace topic name in line 123 in cpp file of pointcloud_to_laserscan.
> 
>     3. The last thing to do is to change your launch file pointcloud_to_laserscan as makanov001 provide in the previous answers.
> 
> 
> Attention. Be sure that you are providing a correct configuration of your lidar in launch.py file. The default one is not fits for any lidar. I mean range_max, range_min and so on.

can you just give your values
i tried many values but none seem to workout

I am getting the same error
```
[pointcloud_to_laserscan_node-2] [INFO] [1741927334.873522169] [pointcloud_to_laserscan]: Message Filter dropping message: frame 'body' at time 1741927334.066 for reason 'discarding message because the queue is full'
```

hermanumrao, 2025-03-14:
nevermind I have solved the issue 
here are the parameters I used:
```
Node(
            package='pointcloud_to_laserscan', executable='pointcloud_to_laserscan_node',
            parameters=[{
                'target_frame': 'body',
                'transform_tolerance': 0.01,
                'min_height': -1.0,
                'max_height': 1.5,
                'angle_min': -3.141592654,  # -M_PI/2
                'angle_max': 3.141592654,  # M_PI/2
                'angle_increment': 0.003141592,  # M_PI/360.0
                'scan_time': 0.3333,
                'range_min': 0.5,
                'range_max': 40.0,
                'use_inf': True,
                'inf_epsilon': 1.0
            }],
            name='pointcloud_to_laserscan'
        )
```